Part-DB/Part-DB-server #1060: Create parts from info provider Digitalkey error

iceyman, 2025-09-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'your partsdb server'
2. Click on 'new parts'
3. Scroll down to 'Create parts from info provider'
4. select digitalkey then lookup item from there

**Expected behavior**
should get a list of items from digital key 

**Screenshots**

[image]

[image]

[image]

**Server Side**
- Part-DB Version: 2.2 [e.g. Part-DB 1.0 (for dev versions commit hash is useful]
- PHP Version: [e.g. PHP 7.4.3]
- Database Server [e.g. MySQL 5.7]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win 10 , Will 11 , ubuntu mate
 - Browser [e.g. chrome, safari] chrome , firefox , edge
 - Version [e.g. 22] 

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

[2025-09-19T05:28:00.200162+00:00] security.DEBUG: Checking support on authenticator. {"firewall_name":"main","authenticator":"Symfony\Component\Security\Http\Authenticator\FormLoginAuthenticator"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:00.200175+00:00] security.DEBUG: Authenticator does not support the request. {"firewall_name":"main","authenticator":"Symfony\Component\Security\Http\Authenticator\FormLoginAuthenticator"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:00.200184+00:00] security.DEBUG: Checking support on authenticator. {"firewall_name":"main","authenticator":"Symfony\Component\Security\Http\Authenticator\RememberMeAuthenticator"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:00.200196+00:00] security.DEBUG: Authenticator does not support the request. {"firewall_name":"main","authenticator":"Symfony\Component\Security\Http\Authenticator\RememberMeAuthenticator"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:00.224967+00:00] request.DEBUG: CSRF validation accepted using both origin and double-submit info. [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:00.226858+00:00] cache.INFO: Lock acquired, now computing item "search_digikey_541-39.0hct-nd" {"key":"search_digikey_541-39.0hct-nd"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.50.1","http_method":"POST","server":"parts.nasstyicey.synology.me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
[2025-09-19T05:28:01.343771+00:00] request.CRITICAL: Uncaught PHP Exception InvalidArgumentException: "Required option not passed: "access_token"" at AccessToken.php line 96 {"exception":"[object] (InvalidArgumentException(code: 0): Required option not passed: "access_token" at /var/www/html/vendor/league/oauth2-client/src/Token/AccessToken.php:96)
[stacktrace]
#0 /var/www/html/vendor/league/oauth2-client/src/Provider/AbstractProvider.php(858): League\OAuth2\Client\Token\AccessToken->__construct()
https://github.com/Part-DB/Part-DB-server/pull/1 /var/www/html/vendor/league/oauth2-client/src/Provider/AbstractProvider.php(653): League\OAuth2\Client\Provider\AbstractProvider->createAccessToken()
https://github.com/Part-DB/Part-DB-server/issues/2 /var/www/html/vendor/knpuniversity/oauth2-client-bundle/src/Client/OAuth2Client.php(123): League\OAuth2\Client\Provider\AbstractProvider->getAccessToken()
https://github.com/Part-DB/Part-DB-server/issues/3 /var/www/html/src/Services/OAuth/OAuthTokenManager.php(110): KnpU\OAuth2ClientBundle\Client\OAuth2Client->refreshAccessToken()
https://github.com/Part-DB/Part-DB-server/issues/4 /var/www/html/src/Services/OAuth/OAuthTokenManager.php(141): App\Services\OAuth\OAuthTokenManager->refreshToken()
https://github.com/Part-DB/Part-DB-server/issues/5 /var/www/html/src/Services/InfoProviderSystem/Providers/DigikeyProvider.php(122): App\Services\OAuth\OAuthTokenManager->getAlwaysValidTokenString()
https://github.com/Part-DB/Part-DB-server/issues/6 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(90): App\Services\InfoProviderSystem\Providers\DigikeyProvider->searchByKeyword()
https://github.com/Part-DB/Part-DB-server/issues/7 /var/www/html/vendor/symfony/cache/LockRegistry.php(111): App\Services\InfoProviderSystem\PartInfoRetriever->{closure:App\Services\InfoProviderSystem\PartInfoRetriever::searchInProvider():86}()
https://github.com/Part-DB/Part-DB-server/issues/8 /var/www/html/vendor/symfony/cache/Traits/ContractsTrait.php(102): Symfony\Component\Cache\LockRegistry::compute()
https://github.com/Part-DB/Part-DB-server/issues/9 /var/www/html/vendor/symfony/cache-contracts/CacheTrait.php(64): Symfony\Component\Cache\Adapter\AbstractAdapter->{closure:Symfony\Component\Cache\Traits\ContractsTrait::doGet():85}()
https://github.com/Part-DB/Part-DB-server/pull/10 /var/www/html/vendor/symfony/cache/Traits/ContractsTrait.php(85): Symfony\Component\Cache\Adapter\AbstractAdapter->contractsGet()
https://github.com/Part-DB/Part-DB-server/pull/11 /var/www/html/vendor/symfony/cache-contracts/CacheTrait.php(30): Symfony\Component\Cache\Adapter\AbstractAdapter->doGet()
https://github.com/Part-DB/Part-DB-server/pull/12 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(86): Symfony\Component\Cache\Adapter\AbstractAdapter->get()
https://github.com/Part-DB/Part-DB-server/pull/13 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(72): App\Services\InfoProviderSystem\PartInfoRetriever->searchInProvider()
https://github.com/Part-DB/Part-DB-server/pull/14 /var/www/html/src/Controller/InfoProviderController.php(172): App\Services\InfoProviderSystem\PartInfoRetriever->searchByKeyword()
https://github.com/Part-DB/Part-DB-server/pull/15 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(183): App\Controller\InfoProviderController->search()
https://github.com/Part-DB/Part-DB-server/pull/16 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\Component\HttpKernel\HttpKernel->handleRaw()
https://github.com/Part-DB/Part-DB-server/pull/17 /var/www/html/vendor/symfony/http-kernel/Kernel.php(182): Symfony\Component\HttpKernel\HttpKernel->handle()
https://github.com/Part-DB/Part-DB-server/pull/18 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\Component\HttpKernel\Kernel->handle()
https://github.com/Part-DB/Part-DB-server/issues/19 /var/www/html/vendor/autoload_runtime.php(29): Symfony\Component\Runtime\Runner\Symfony\HttpKernelRunner->run()
https://github.com/Part-DB/Part-DB-server/issues/20 /var/www/html/public/index.php(5): require_once('...')
https://github.com/Part-DB/Part-DB-server/pull/21 {main}
[... cut ...] .me","referrer":"[https://parts.nasstyicey.synology.me/en/tools/info_providers/search"}](https://parts.nasstyicey.synology.me/en/tools/info_providers/search%22%7D)
192.168.50.3 - - [19/Sep/2025:05:28:00 +0000] "POST /en/tools/info_providers/search HTTP/1.1" 500 4514 "https://parts.nasstyicey.synology.me/en/tools/info_providers/search" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/140.0.0.0 Safari/537.36"

19/Sep/2025:05:28:00 +0000 "POST /index.php" 500

jbtronics, 2025-09-29:
You probably need to refresh the digikey oauth token, as these get invalidated after a while. You can click "Connect oauth" in the provider list...

iceyman, 2025-09-30:
> You probably need to refresh the digikey oauth token, as these get invalidated after a while. You can click "Connect oauth" in the provider list...

this wored thanks didnt think i needed to do that more than once should set up something to make it auto do it once every say 7 days